luni64/EncoderTool #14: incompatibility with LCD_I2C

mattrcampbell, 2022-04-22:
I'm having an issue with an incompatibility with the LCD_I2C library I can't quite put my finger on and would appreciate any insight if anyone has any....

It issue is that when the EncoderTool library is in use, the lcd.printf(...) method stops doing anything.  lcd.print(...) still works.  It does not interrupt code execution, it just has no impact.  Take out the EncoderTool lib and it works again.

Using https://github.com/blackhack/LCD_I2C.
Hardware is teensy 3.6

Sample code that exhibits the issue:
```
#include <Arduino.h>
#define ENCODER_OPTIMIZE_INTERRUPTS

#include <EncoderTool.h>
#include <LCD_I2C.h>

using namespace EncoderTool;
Encoder enc1;
LCD_I2C lcd(0x27,40,2);
long lastPos;
int i = 0;

void setup() {
  lcd.begin();
  lcd.backlight();
  lcd.clear();
  enc1.begin(26, 25);
}

void loop() {
  lcd.setCursor(0, 0);
  lcd.print(i % 10);
  int32_t wheelin = enc1.getValue();
  if (wheelin != lastPos) {
    lcd.setCursor(0, 1);
    char l[50];
    sprintf(l, "%ld", wheelin);
    lcd.print(l);  // THIS WORKS

    lcd.setCursor(10, 1);
    lcd.printf("%d", wheelin); // THIS DOES NOT
    lastPos = wheelin;
  }
  i++;
}

```

Any thoughts?

mattrcampbell, 2022-04-22:
Ah-ha!  Found the issue, it is caused by this commit: https://github.com/luni64/EncoderTool/commit/87d902abb2bea7769c1d204a4f7e297d4a5ac13c
Removing that code makes it work as expected.  Unsure of the right solution besides editing the library though.

luni64, 2022-04-22:
Thats weird.  Here the corresponding thread from the forum https://forum.pjrc.com/threads/65469-EncoderTool-Problems which triggerd the 'fix'. 
I'll check if one of the more recent Teensyduino versions fixed the original problem so that the definition of _write is not needed anymore.

luni64, 2022-04-22:
Can you please check if replacing the code in `src/encoderbase.cpp` line 189ff by 
```c++
#if defined(KINETISK)
extern "C" {
int _write(int file, char* ptr, int len)
{
    ((class Print*)file)->write((uint8_t*)ptr, len);
    return len;
}
}
#endif
```
fixes the issue for you (it does for me...)

mattrcampbell, 2022-04-22:
Yes, that seems to work!

luni64, 2022-04-22:
Thanks for testing, I'll do some further checks and push it to the master branch on the weekend. 

In case you don't know, you can use valueChanged() to simplify your code: 
```c++
void loop() {
   lcd.setCursor(0, 0);
   lcd.println(i % 10);
  
   if(enc1.valueChanged()) {
     int32_t wheelin = enc1.getValue();
     lcd.setCursor(0, 1);
     lcd.printf("%d\n", wheelin);
   }
   i++;
}
```
Also `ENCODER_OPTIMIZE_INTERRUPTS` is not used by the EncoderTool. Seems to be a setting for the stock Encoder library.

mattrcampbell, 2022-04-22:
Yeah, `ENCODER_OPTIMIZE_INTERRUPTS` was just leftover from me testing that the stock lib worked and I forgot to remove it from that test code.  Great tip on the `valueChanged(),` thanks!

luni64, 2022-04-24:
fixed  by [385dc5b](https://github.com/luni64/EncoderTool/commit/385dc5b8c25e4b2582321d6e869ed12ce4e42e5e)

closed 2022-04-24 by luni64