luni64/EncoderTool #21: Issue with TeensyLC

bimac, 2022-07-07:
Hi luni64,

I have a strange issue with EncoderTool on TeensyLC.

Here's the MWE:
``` cpp
#include <EncoderTool.h>

using namespace EncoderTool;
PolledEncoder enc;

void setup() {
  Serial.begin(9600);
  enc.begin(5, 6, 4, CountMode::half, INPUT_PULLUP);
}

void loop() {
  enc.tick();
  if(enc.valueChanged())
    Serial.println(enc.getValue());
}
```

On Teensy 3.2 this works perfectly fine: each detent will increment/decrement by 1. On TeensyLC, however, it will keep incrementing like crazy (without turning the encoder) with each iteration of ```loop()``` - but _only for every other detent_. I'm using [this encoder](https://www.digikey.de/de/products/detail/bourns-inc/PEC11L-4225F-S0015/4699180). Is there anything that comes to mind?

Cheers!

luni64, 2022-07-07:
This looks weird. Unfortunately I'm currently not at home and don't have a LC with me. I'll have a look on Sunday.

bimac, 2022-07-07:
Thanks a lot!

luni64, 2022-07-10:
I can reproduce this. It looks like there is a bug in the direct read code for the T-LC. I'll have a closer look later. As a quick workaround you can do the following: 

Change line 46 in the library file cores.h as shown here: 
[image]

This disables the direct read code for the T-LC so that it falls back to a standard digitalRead/digitalWrite which works here. 
Can you please check if this works for you as well?

bimac, 2022-07-10:
Thanks for looking into this! I can confirm that the workaround helps.

luni64, 2022-07-10:
Great, I'll look for a permanent solution asap

closed 2022-07-22 by luni64

bimac, 2022-07-22:
Thanks for the fix Lutz! It works as expected now.

luni64, 2022-07-22:
Great, let me know if you find something else

PR #23 (2022-07-22): Fix directRead/Write for TeensyLC
fixes #21