luni64/EncoderTool #22: Library impossible to include from two cpp files

maxgerhardt, 2022-07-19:
See [topic](https://community.platformio.org/t/multiple-definition-of-error-teensy-4-1/28842).

Due to the `Encoder.h` class inlining the implementation, two files using the `Encoder` after doig `#include <EncoderTool.h>` will result in a double definition error.

Minimal reproduction with PlatformIO:
`platformio.ini`
```ini
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
lib_deps = https://github.com/luni64/EncoderTool.git
```

with `src/main.cpp`
```cpp
#include <Arduino.h>
#include <EncoderTool.h>

using namespace EncoderTool;
Encoder enc1;
extern Encoder enc2;

void setup() {
    enc1.begin(1, 2);
    enc2.begin(3, 4);
}

void loop() {}
```
and `src/helper.cpp`
```cpp
#include <Arduino.h>
#include <EncoderTool.h>

using namespace EncoderTool;
Encoder enc2;

void some_func() { }
```
leads to

```
.pio\build\teensy41\src\main.cpp.o: In function `EncoderTool::Encoder::Encoder()':
main.cpp:(.text._ZN11EncoderTool7EncoderC2Ev+0x0): multiple definition of `EncoderTool::Encoder::Encoder()'
.pio\build\teensy41\src\helper.cpp.o:helper.cpp:(.text._ZN11EncoderTool7EncoderC2Ev+0x0): first defined here
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.50401.190816/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions        
.pio\build\teensy41\src\main.cpp.o: In function `EncoderTool::Encoder::Encoder()':
main.cpp:(.text._ZN11EncoderTool7EncoderC2Ev+0x0): multiple definition of `EncoderTool::Encoder::Encoder()'
.pio\build\teensy41\src\helper.cpp.o:helper.cpp:(.text._ZN11EncoderTool7EncoderC2Ev+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```

maxgerhardt, 2022-07-19:
https://github.com/maxgerhardt/EncoderTool/commit/88d7f36ba95f2bace6af81b59e3011dcaa7f18ae would solve this issue but get rid of the `inline` definitions.

luni64, 2022-07-20:
fixed in v3.0.3

closed 2022-07-20 by maxgerhardt