luni64/EncoderTool #31: Push button support for Multiplexed encoders

Cinezaster, 2023-02-08:
Am I correct to assume their is no push button support for Multiplexed encoders?
Is it in the pipeline?

luni64, 2023-02-09:
Actually there is support for push button support. See e.g.  (line 15): https://github.com/luni64/EncoderTool/blob/decf3e781214c33aa57cc1fd14a2df741ed97e7d/src/Multiplexed/EncPlex74165.h#L11-L23
Which mulitplexing scheme are you aiming for?

Cinezaster, 2023-02-09:
I'm using the 4051, I did not look into the other multiplexer codes. 
https://github.com/luni64/EncoderTool/blob/master/src/Multiplexed/EncPlex4051.h 
I might implement it myself. 
My project has way more IO's to deal with so, I'm not sure if I'm going to keep your awesome library. In any case thx for your project and your quick response.

luni64, 2023-02-09:
In case you add the push button (which should be simple) to the 4051 example I'd be happy to update it. 

> I'm not sure if I'm going to keep your awesome library

Thanks! Let me know if you need any help or "decission support"  :-)

serlecu, 2023-10-06:
Looking inside src, it looks like you only need to add an extra argument to the `EncPlex74165` :

https://github.com/luni64/EncoderTool/blob/master/src/Multiplexed/EncPlex74165.h
```
inline EncPlex74165_tpl(unsigned nrOfEncoders, unsigned pinLD, unsigned pinCLK, unsigned pinA, unsigned pinB, unsigned pinBtn = -1);
```
So you give the QH pin of the third encoder (pushButtons) as`pinBtn` argument value.

Then you could use set the callbacks with `attachButtonCallback(encBtnCallback_t)` as it appears in [__EncoderBase.h__ line 35](https://github.com/luni64/EncoderTool/blob/master/src/EncoderBase.h) what will return `(uint_fast8_t channel, int_fast8_t state)`to your callback function as it appears in [EncPlexBase.h line 17](https://github.com/luni64/EncoderTool/blob/master/src/Multiplexed/EncPlexBase.h)

it is also supposed to be analogous to lines 41 and 41 of the [74165 Multiplexed example](https://github.com/luni64/EncoderTool/blob/master/examples/3_callbacks/mplexCallbacks/mplexCallbacks.ino):
```
 encoders[0].attachCallback(onFirstEncoder); // standard callback
 encoders.attachCallback(onAnyEncoder);
```
I might guess it turns into:
```
encoders[0].attachButtonCallback(onFirstEncoderBtn); // standard callback
encoders.attachButtonCallback(onAnyEncoderBtn);
```

I'll test it soon

serlecu, 2023-10-06:
> **EncoderBase.h** line 35

@hakonbraga 
just tested and somehow Arduino IDE cant find `attachButtonCallback()` for using it with the whole array of encoders, but you can still assign individual callbacks like this:
```
constexpr unsigned QH_A = 2;   // output pin QH of shift register A
constexpr unsigned QH_B = 3;   // output pin QH of shift register B
constexpr unsigned QH_S = 4;   // output pin QH of shift register S
constexpr unsigned pinLD = 5;  // load pin for all shift registers)
constexpr unsigned pinCLK = 6; // clock pin for all shift registers

EncPlex74165 encoders(encoderCount, pinLD, pinCLK, QH_A, QH_B, QH_S);

void onFirstEncoderBtn(int state){}
void onSecondEncoderBtn(int state){}

void setup() {
...
encoders[0].attachButtonCallback(onFirstEncoderBtn);
encoders[1].attachButtonCallback(onSecondEncoderBtn);
...
}

void loop() {
    encoders.tick();
}
```

bsom, 2024-12-13:
Could really use this in the 4051 portion of this library... :-)

luni64, 2024-12-13:
I can take a look at it over the weekend—it shouldn't be too difficult. However, I’d appreciate it if you could thoroughly test it, as I don’t have much time for testing right now.

bsom, 2025-01-14:
I can certainly test it as I'm in the middle of designing a device that requires it. :-)

luni64, 2025-01-14:
Oh, since you didn't respond I thought you are not interested anymore. I'll try to come up with a solution the next days.

luni64, 2025-01-15:
I uploaded an experimental version to the branch "AddMultiplexedButton_4051".  I currently can't test it, so I hope it works. At least it compiles :-) 

Here some test code (also untested). It uses three 4051 sharing the same address pins S0..S2. Two of the chips handle the phases A/B and the third handles the buttons. 
```c++
#include "EncoderTool.h"
using namespace EncoderTool;

constexpr unsigned encoderCount = 8; // number of attached encoders

constexpr unsigned S0 = 0;      // address pin 0
constexpr unsigned S1 = 1;      //...
constexpr unsigned S2 = 2;      // address pin 2
constexpr unsigned SIG_A = 3;   // output pin SIG of multiplexer phase A
constexpr unsigned SIG_B = 4;   // output pin SIG of multiplexer phase B
constexpr unsigned SIG_BTN = 5; // output pin SIG of multiplexer button
                                // breakout:  https://www.sparkfun.com/products/13906
                                // datasheet: https://www.ti.com/lit/gpn/CD74HC4051

EncPlex4051 encoders(encoderCount, S0, S1, S2, SIG_A, SIG_B, SIG_BTN);

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
    encoders.begin();
}

unsigned t0 = 0;

void loop()
{
    encoders.tick();

    for (unsigned i = 0; i < encoderCount; i++)
    {
        if(encoders[i].valueChanged())
            Serial.printf("E%d: %d\n", i, encoders[i].getValue());
        if(encoders[i].buttonChanged())
            Serial.printf("B%d: %d\n", i, encoders[i].getButton());;
    }
}
```
Please let me know if it works

bsom, 2025-01-15:
You are correct, it compiles.  Now I just need to figure out if it works with callbacks for buttons. :-)

luni64, 2025-01-15:
If you want to attach different callbacks for the various buttons you can use `encoders[i].attachButtonCallback`. I didn't implement the combined callback for all buttons since this would require changes in the base classes which I'd rather not do without testing possibiltiy.  However, you can easily achive the same effect by using lambdas. 
```c++
#include "EncoderTool.h"
using namespace EncoderTool;

constexpr unsigned encoderCount = 8; // number of attached encoders

constexpr unsigned S0 = 0;      // address pin 0
constexpr unsigned S1 = 1;      //...
constexpr unsigned S2 = 2;      // address pin 2
constexpr unsigned SIG_A = 3;   // output pin SIG of multiplexer phase A
constexpr unsigned SIG_B = 4;   // output pin SIG of multiplexer phase B
constexpr unsigned SIG_BTN = 5; // output pin SIG of multiplexer button
                                // breakout:  https://www.sparkfun.com/products/13906
                                // datasheet: https://www.ti.com/lit/gpn/CD74HC4051

EncPlex4051 encoders(encoderCount, S0, S1, S2, SIG_A, SIG_B, SIG_BTN);

void callback(int encNr, int state)  // will be called for each button
{
    Serial.printf("Button: %d, State: %d", encNr, state);
}

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);

    for (unsigned i = 0; i < encoderCount; i++)
    {
        encoders[i].attachButtonCallback([i](int state) { callback(i, state); });
    }

    encoders.begin();
}

unsigned t0 = 0;

void loop()
{
    encoders.tick();
}

```

bsom, 2025-01-16:
Well, it compiles.  So, step one is good! :-)
I'm about to test it on actual hardware.  I'll check back in when I that is done.
Thanks again for your quick turnaround on this.

bsom, 2025-01-16:
Just tested it on real hardware and it is working well.  Didn't need to debounce the presses either, which I was concerned might be an issue.

I made an individual callback for each of eight encoder buttons and it all worked swimmingly well.

Thanks so much!!!

luni64, 2025-01-16:
Great! 
I'll wait a few days and if I don't hear anything negative from you I'll merge it into the main branch. 
Thanks for testing

PS: the buttons are debounced internally (I'm using the bounce2 lib from Thomas Frederic https://github.com/thomasfredericks/Bounce2)

ensonic, 2025-03-22:
I've just tested it. Works great. If you don't mind, it would be great to also let people specify the inpt_mode (see https://github.com/luni64/EncoderTool/discussions/44).

Vladistone, 2025-05-02:
I apologize if this discussion was closed. But I want to know the author's opinion about the compatibility of his code with another MUX connection scheme. Or give recommendations on reorganizing the code recoding according to my scheme. because I am also considering upgrading an already completed device with 32 encoders, adding another 32 BTNs. The device itself is made on the principle of connecting MUX in this way: encoder #1 to #4 to 1 MUX#1 and so on... (see the schematic diagram in the appendix)

[image]

[image]