luni64/EncoderTool #36: MUX with single 74HC165 and 4 Encoders

serlecu, 2023-10-05:
Hi,

I'm unsuccessfully trying to implement the example of multiplexing encoders with a 74HC165 on an Arduino Micro, but using only one multiplexer (also setting `encoderCount` to 4).

Turning the first encoder (pins 6 & 7 of the IC) I'm getting the following readings:
```
Encoder: 0 value: 1385 delta: -1
Encoder: 1 value: 318 delta: 1
Encoder: 0 value: 1386 delta: 1
Encoder: 1 value: 317 delta: -1
Encoder: 0 value: 1385 delta: -1
Encoder: 1 value: 316 delta: -1
Encoder: 0 value: 1384 delta: -1
Encoder: 1 value: 315 delta: -1
Encoder: 0 value: 1383 delta: -1
Encoder: 0 value: 1382 delta: -1
Encoder: 0 value: 1383 delta: 1
```
Turning any other Encoder has no effect.

All encoders are routed directly to the IC as they are boarded with pull-up resistors.

I also tried with two ICs, because I don't fell 100% shure I can set `QH_B = -1` as it is set by default for the switches' IC. Got any luck either.

Am I doing anything really silly?

Thx in advance.

luni64, 2023-10-06:
>I'm unsuccessfully trying to implement the example of multiplexing encoders with a 74HC165 on an Arduino Micro, but using only one multiplexer (also setting encoderCount to 4).

That won't work with the example code, it expects the A and B signals on different encoders. There is a schematic in the repo showing how to connect the encoders. https://github.com/luni64/EncoderTool/tree/master/Resources/Extras#Multiplexerwith74HC165
If you want to connect your 4 Encoders to one 74165 you need to adapt the reading code see here how this is done for another multiplexer. Translation to a '165 should be straight forward https://github.com/luni64/EncoderTool/tree/master/examples/2_multiplexing/multiplexed_MCP23S17

serlecu, 2023-10-06:
Ah! Sorry @luni64, I didn't realise that the IC's labels in your schematic were corresponding to:
A1 -> Pin A of Encoder 1
B1 -> Pin B of Encoder 1
A2 -> Pin A of Encoder 2
And so on...

All clear!!
Thanks a lot for your quick reply and this library

closed 2023-10-06 by serlecu