luni64/EncoderTool #40: Some schematic explanation needed

SpacewormSera, 2023-12-08:
Hi! 

Thanks for such a nice project! 

I want to use a Raspberry pi Pico board as a dedicated device for encoders only. The board has enough pins to use 8 encoders connected directly to it, so I've chosen the direct mux option. The encoders I wanna use are EC11 with 2 encoders pins, one switch and the ground.

Here is the problem:  this schematic uses some other type of encoders: https://github.com/luni64/EncoderTool/blob/master/Resources/Extras/Boards/MATRIX/documentation/schematic.jpg  

so what is the wiring for the EC11 type? 
Thanks!

luni64, 2023-12-08:
>Here is the problem: this schematic uses some other type of encoders

Actually it uses standard encoders. 

[image] 

I just didn't want to draw all those decoupling diodes and therefore defined a schematic symbol containing them.  If you look at the board you can see those diodes on the back side: 
[image], 

So, just connect the diodes to the pins of the encoders as shown in the symbol and you are good to go. 

> And also - can you give some hints about the user experience, are there some missing readings, for example if i turn 2 encoders simultaneously, or which schematic runs the most smoothly?

Of course this depends on what else you want to do with the board. As a rule of thumb your code needs to update the encoders (call tick()) with more than 2-3 kHz to work without issues. If you can not guarantee that in `loop()` you might consider calling `tick()` from a timer interrupt,  

> or which schematic runs the most smoothly?

The most easy to use scheme for only 8 encoders is using a 4051 (see here: https://github.com/luni64/EncoderTool/blob/master/Resources/Extras/Boards/MPX_4051/MPX_4051_Schematic.pdf and here: https://github.com/luni64/EncoderTool/blob/master/src/Multiplexed/EncPlex4051.h)

Or, since you wrote that you have enough pins for the encoders you could connect them directly. 

I never tested the code with a Pico. Let me know if works or if you need some help.

SpacewormSera, 2023-12-11:
Thanks for the help, I'll post my results here then :)

closed 2024-07-28 by luni64